execuc/u2if #2: Interested in adding other RP2040 boards?

caternuson, 2021-05-24:
Hey @execuc , this isn't an issue.

First, thanks for writing this cool firmware. I did the work to add the additional support needed to allow using the Pico w/ u2if firmware from a PC via the Blinka library:
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-raspberry-pi-pico
It all worked really well!

I've also done some simple modifications to allow adding other RP2040 based boards:
https://github.com/adafruit/u2if/pull/1
which generally came down to a pin re-mapping. Would you be interested in a pull request to add that capability here?

Also - curious what build environment you used for your release artifacts? I was originally using `gcc-arm-none-eabi-10-2020-q4-major` to build for the PR linked above and had issues with it creating an infinite loop inside `StreamedInterface::streamRxRead`. Eventually tried just switching to `gcc-arm-none-eabi-7-2018-q2-update` which worked as expected. Really weird issue.

execuc, 2021-05-25:
Hello !

I am very honored that this firmware has been integrated into Blinka for the CircuitPython. Adafruit products are very good and you make DIY electronics progress a lot with your github repositories.

Although the pico is very flexible in terms of pin mapping, I wanted to freeze them and match them to the PCB that I made at the same time. It was for simplicity but I think it was a mistake.
Anyway, it is a good idea to be able to add other RP2040 based cards and therefore, I will gladly accept a pull request, thank you!

This infinite loop problem is very strange. For my part, I am developing on a debian 10 which provides gcc-arm-none-eabi in version "15: 7-2018-q2-6". If I have a chance to upgrade gcc, I will try to find the problem, thanks for the information!

caternuson, 2021-05-25:
May have fixed the weird compiler issue - or at least tweaked the code to get things working. See:
https://github.com/adafruit/u2if/pull/2
Still odd that the compiler would produce what it did from the original code. Those code changes are included in #3.

execuc, 2021-06-04:
Thanks for your pull request.
I made a new release (v0.4) with the firmware of the 4 Adafruit boards.

I also added the UART1 interface because I saw that the QT Py card was wired to UART1 and not UART0. So there may be a modification in Blinka in order to handle both uart (as you made it for SPI and I2C).

caternuson, 2021-06-14:
woot! thanks!

closed 2021-06-14 by caternuson